rpi-ws281x/rpi-ws281x-python #6: examples crash

iZhangHui, 2018-06-22:
1. Hardware: Raspberry Pi 3B

2. Library: Installing collected packages: rpi-ws281x
             Successfully installed rpi-ws281x-3.1.0

3. OS: Raspbian Stretch

4. kernel: pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.14.34-v7+ #1110 SMP Mon Apr 16 15:18:51 BST 2018 armv7l GNU/Linux

5. error information:
python3 strandtest.py

Traceback (most recent call last):
  File "/home/pi/rpi-ws281x-python/examples/strandtest.py", line 91, in <module>
    strip.begin()
  File "/home/pi/.local/lib/python3.5/site-packages/rpi_ws281x/rpi_ws281x.py", line 124, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -5 (mmap() failed)

burckaan, 2018-07-02:
I am having the same problem, is there a solution coming up?

iZhangHui, 2018-07-02:
@burckaan With a lot of experiments, I found a solution as below:

**sudo** pip3 install rpi-ws281x
**sudo** python3  strandtest.py

It works for me now.:)

IsaiahGrace, 2020-05-01:
I've been experiencing this issue too. I want to control three separate  led strips at once. I'm using pins 10 (SPI mode),13, and 18 (PWM mode). I've used sudo to run my python script as iZhangHui suggests, but I think it should be possible to run the script without using sudo. I did some digging and it looks like the function call that fails (mmap) is attempting to map memory which the user does not have the permissions to do. 
The keys here are that the user must have permissions to access `/dev/mem` AND (I think) must have `CAP_SYS_RAWIO`. Now I'm not entirely sure how to do those things, but this is what I've figures out so far. Perhaps there is a way to add an exception for this specific program? I think there should be a solution that doesn't give blanket memory access to the pi user all the time.
Sorry to dig up a dead bug report like this, but hopefully someone who knows a little more than I can think up a clever and elegant solution.

patrickscottart, 2021-02-23:
I am having the same problem trying to control ws2815 string.  Only now, the problem originates on "line 131, in begin . . . (mmap() failed)"

And the solution above: sudo python3 strandtest.py  brings error "No such file or directory"

FranzSperling, 2022-07-13:
`sudo python3 -m pip install --force-reinstall adafruit-blinka` worked for me.

closed 2023-05-16 by Gadgetoid