rpi-ws281x/rpi-ws281x-python #8: backsupport for adafruit-circuitpython-neopixel

ladyada, 2018-08-27:
hey folks, a few years ago we added a 'neopixel' module to this repo, which is awesome but has ended up name-colliding with our circuitpython neopixel library. because of import order, if we install `rpi_ws281x`, it will never use adafruit-circuitpython-neopixel's neopixel.py file. which is life and i can live with it:)

to 'fix' we'd like to add a new class to [this repo's neopixel module](https://github.com/rpi-ws281x/rpi-ws281x-python/tree/master/library/neopixel) that will essentially be a [copy-n-paste of this file](https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/blob/master/neopixel.py), to create a circuitpython-api-compatible neopixel object. [this will let us complete our quest to provide a unified API for CPython/circuitpython hardware projects](https://learn.adafruit.com/circuitpython-on-raspberrypi-linux)

It shouldn't affect anything on your end cause it will be a completely new class, but let me know whatcha think! I can do the PR if it sounds good to you :)

Gadgetoid, 2018-08-28:
Since the `neopixel` name is your brand-name (registered trademark IIRC?), and since it was only kept in this library as a compatibility shim after I moved for the more generic (but still not quite all-encompassing) `rpi_ws281x` I'd be happy to advocate for the dropping of `neopixel` altogether from this library, so that you can use the module name without conflicts or needing to worry about this repository in future.

A major version bump on the associated release of `rpi_ws281x-python` and simple instructions to end-users should get them migrated from using "neopixel" to "rpi_ws821x" which are presently totally interchangeable. Either way our users (at least those still `import neopixel`) are going to have to make that change.

Would this work for you?

Gadgetoid, 2018-08-28:
In further reading of your solution I understand why it wouldn't affect our users. Still I suspect it would cause some amount of headache with future maintenance since a copy & paste of the class would have to be maintained in two places.

I'm happy to take the headache (I suspect it wont be particularly problematic) of moving users from `import neopixel` to `import rpi_ws281x` so that you can have your module name to yourself without any quirks or complications.

After all, the library is called `rpi-281x-python` and it makes little to no sense to `import neopixel` from a package with a completely different name.

As much as I can't stop calling everything "neopixels" by habit (and I wholeheartedly recognise the Adafruitian origins of the Python class upon which we have built), the nomenclature clash in this library has bugged me for years :grin: (I felt shipping this on PyPi as "neopixel" would have been crossing a line)

Gadgetoid, 2018-08-28:
Oh and a heads up (ugh sorry for the wall of replies), regardless of the outcome of this issue- I think this learn article is quite out of date: https://learn.adafruit.com/neopixels-on-raspberry-pi?view=all and could replace a vast majority of the install fiddling with our shipped Python wheels:

```
sudo pip install rpi_ws281x
```

Since over on https://github.com/jgarff/rpi_ws281x we're often fielding issues from users trying to build the crusty old Python bindings lingering in that repository from source when they don't need to:

https://github.com/jgarff/rpi_ws281x/issues/307
https://github.com/jgarff/rpi_ws281x/issues/299
https://github.com/jgarff/rpi_ws281x/issues/291
https://github.com/jgarff/rpi_ws281x/issues/290

ladyada, 2018-08-28:
whew! ok yeah we're ok with removing neopixel from this library completely, and we'll update our code/guide as well! as mentioned, we'll be showing how to use the circuitpython api anyhow :)
let us know when you've yanked & bumped pypi!

Gadgetoid, 2018-08-29:
You may forge ahead! Godspeed.

https://github.com/rpi-ws281x/rpi-ws281x-python/releases/tag/v4.0.0

ladyada, 2018-08-29:
thankx!

closed 2018-08-29 by ladyada