rpi-ws281x/rpi-ws281x-python #13: No support for RGBW?

WyldeLaRue, 2019-01-11:
I'm updating my old project from the python wrapper at https://github.com/rpi-ws281x. I got an error because Color only takes 3 values as input. After looking through the code, it seems like there is no mention at all of 32 bit color. 

Are there any plans to fix this? It was supported before, that was one of the reasons I ended up using this library.

Phierdomenico, 2019-02-03:
Same here. I'm very interested to this issue.

Jianan-Li, 2019-04-20:
Confirmed that the python wrapper currently doesn't support RGBW. Please fix this.

rpi_ws281x version: 4.1.0
  
`pi@raspberrypi:~/Documents/Projects/ws281x_examples $ sudo python3 SK6812_strandtest.py 
Press Ctrl-C to quit.
Traceback (most recent call last):
  File "SK6812_strandtest.py", line 94, in <module>
    colorWipe(strip, Color(0, 0, 0, 255))  # White wipe
TypeError: Color() takes 3 positional arguments but 4 were given
`

closed 2023-05-16 by Gadgetoid