rpi-ws281x/rpi-ws281x-python #14: RuntimeError when using within Docker container

charliesneath, 2019-02-08:
When running my program within a Docker container, I get the following error, which I believe is related to `rpi_ws281xs`'s ability to interact with the hardware from the Docker container:

```
RuntimeError: ws2811_init failed with code -9 (Failed to create mailbox device)
```

Is there a way to make this library work within a Docker container?

Also reported at https://github.com/jgarff/rpi_ws281x/issues/351.

charliesneath, 2019-02-12:
@Gadgetoid Any ideas on this error? Is there additional information I can provide to help debug/evaluate?

sylvek, 2019-02-12:
@charliesneath Do you run your docker with user 0, group 0 and privileged option?

charliesneath, 2019-02-13:
@sylvek I wasn't setting this explicitly, but I will look into doing so. Do you have any examples?

sylvek, 2019-02-13:
`docker run --privileged …` should work.

charliesneath, 2019-02-19:
@sylvek That did the trick, thank you.

closed 2019-02-19 by charliesneath

janblumenkamp, 2024-02-15:
Is there any way to get this to work without the --privileged mode? What devices should be mapped to the docker container?

Gadgetoid, 2024-02-16:
> Is there any way to get this to work without the --privileged mode? What devices should be mapped to the docker container?

This would be better asked on the repo for the underlying library, since the Python bindings don't affect its limitations - https://github.com/jgarff/rpi_ws281x

However, I suspect the answer is no. But you could paint what I know about Docker on the head of a pin with a firehose.