rpi-ws281x/rpi-ws281x-python #17: ws2811 render failed with code -14 when pixels is > 339

Fisherman86179, 2019-05-22:
Hello,

**Issue 1**
When running the strandtest.py w/ more than 339 WS2812B pixels (as a strip) in SPI mode I get following error:

Traceback (most recent call last):
File "/home/pi/rpi_ws281x-master/python/examples/strandtest_new.py", line 102, in 
colorWipe(strip, Color(255, 0, 0)) # Red wipe
File "/home/pi/rpi_ws281x-master/python/examples/strandtest_new.py", line 30, in colorWipe
strip.show()
File "/usr/local/lib/python3.5/dist-packages/rpi_ws281x/rpi_ws281x.py", line 131, in show
raise RuntimeError('ws2811_render failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_render failed with code -14 (SPI transfer error)

With less pixels it works fine. What is driving this limit?
Is there a way to increase the pixel number to 512 and could you please point me in the right direction, if so?

Pi 3B+, Fresh Stretch and your libs installed two days ago, Python3.
Python modules NeoPixel and rpi-ws281x are both installed? Is that OK?

.boot/cmdline.text ammended by: spidev.bufsiz=32768

**Issue 2**
I noticed that the light intensity of blue is low. Red and green are bright, but not so blue. 
De facto running (modified) strandtext.py:

[image]

reveals that 100% Blue is not brighter as Red or Green when set to less 50%.

I have driven this led string using an Arduino and I can confirm that there is "more of blue" to come out, i.e. the strip is OK.

Thanks for your time,
Juergen

PS: I posted this also to Jeremy Garff. The issue is the same, weather I use Neopixel or rpi-WS281x

Fisherman86179, 2019-05-26:
Ladies and Gents,

Issue 2 is not an issue but my fault, I would say.

Indeed, blue is not very bright when compared to e.g. green, but this seems not to be a software issue at all. I digged w/ the scope in the data sent and can confirm that all 8 bits for blue are "1".
Furthermore I measured the current for blue, green and red (72 pixels) and they are the same: 0,94A at a voltage loss of 0,338V across shunt. So the current consumption of fore each led (RGB) the same.

Digging in the data sheet of the WS2812B finally revealed the issue:

[image]

As shown, blue emmission is poor, in w.c. it can be 180mcd for B and 720 mcd for G. A significant difference.

So, my fault, sorry, issue closed. I post this message just for everybody else trapping in this issue.

A calibration routine could help, if really smooth colors are needed.

**Issue 1 is still persistent.**
I found folloing statement re Kernel and SPI:
https://github.com/raspberrypi/linux/issues/2165

Is this still true?

Gadgetoid, 2019-06-28:
Looks like this is a dupe of https://github.com/jgarff/rpi_ws281x/issues - which is the correct place to report issues with the underlying rpi_ws281x library. This GitHub repository only deals with the Pythonw rapper, and I'm pretty confident your issue is with the C-code with which I am not particularly well versed.

closed 2019-06-28 by Gadgetoid