rpi-ws281x/rpi-ws281x-python #19: Can't use rpi_ws281x together with second PWM channel

olivierswinkels, 2019-07-06:
I'm using rpi_ws281x (via python) in PWM mode on PWM0 (GPIO 12) and this works fine standalone. 
However when I try to initialize another PWM channel on PWM1 (GPIO 13) (Using either wiringpi or pigpio) the rpi_ws281x signal stops.  (I'm using rpi_ws281x 4.2.2 on a Raspberry Pi 3B+ running Raspbian.)

Is this a known issue, or is there another way to use the rpi_ws281x library together with a 'raw' hardware PWM signal on the other channel?

dallen98, 2020-10-20:
Did you get a solution to this as I want the same thing kind of, I want to change all availabe GPIO pins to pwm using pigpio and then my question becomes does the rpi_ws281x work on pigpio virtual pins ?

I have repeated the question here sort of...
https://github.com/jgarff/rpi_ws281x/issues/419

Gadgetoid, 2020-10-20:
If a pin does not have PWM hardware physically connected to it, then it will not be able to output a hardware PWM signal. rpi_ws281x *requires* hardware PWM.

pigpio - IIRC - just uses software tricks on non-hardware PWM pins, if not all pins.

If you want to send the same PWM signal to multiple strips of LEDs... then just wire them to the same pin.

dallen98, 2020-10-20:
"If you want to send the same PWM signal to multiple strips of LEDs... then just wire them to the same pin."
this is a large led matrix i,m trying to build hence my need for a large number of output pins outputting different signals at the same time so from your reply it suggests that what i,m trying to do is impossible NB i,m confined to pin 18 and 19 only ? for unique signal outputs, thanks

Gadgetoid, 2020-10-20:
Indeed, it's impossible short of some programming wizard coming along and figuring out this - https://github.com/jgarff/rpi_ws281x/issues/417#

At best you can get four simultaneous outputs - if I remember correctly - using PCM and PWM.

mfaccin, 2020-11-19:
Hi @Gadgetoid. I have an application running in parallel using the pwm1 and accessing the registers directly.

Configuring the rpi_ws281x to use only the pwm0 / GPIO18 pin mess up with the other service. Is this limitation you described here?

> Edit: Note the above statement is unrelated to the use of PWM0 alongside PWM1 for different purposes. I do not know exactly why it's not possible to use these simultaneously with different libraries, but I'd very strongly suspect that either rpi_ws281x or whatever othere library is attempting to output PWM is setting up pins or features it shouldn't be. Or that certain shared hardware peripherals or techniques make using both together impossible.

d8ahazard, 2021-01-06:
> Indeed, it's impossible short of some programming wizard coming along and figuring out this - [jgarff/rpi_ws281x#417](https://github.com/jgarff/rpi_ws281x/issues/417)
> 
> At best you can get four simultaneous outputs - if I remember correctly - using PCM and PWM.

Hi, could you please elaborate on how it would be possible to get four simultaneous outputs using a Raspi 4B?

I would assume that's using GPIO 18/19 for PWM, and 21 for PCM...but what is the other GPIO? 31 is not available on the 4b.

dallen98, 2021-01-06:
there are only 2 unique outputs on rpi 18/19 the other 2 pwm pins repeat the data coming out of 18 and 19

sporubcan, 2022-07-25:
Hello, I have the same problem.
I have FAN PWM control connected to GPIO13 and NeoPixel to GPIO12.
Both are working standalone well.
But when I start FAN PWM on GPIO13 (channel 1), it interferes NeoPixel on GPIO12 (channel 0).
Did you find the solution please?
Thank you in advance!