rpi-ws281x/rpi-ws281x-python #24: Runtime error on Pi 3 A+

jonodav, 2019-08-03:
Running the examples or any other python scripts results in an error code -3 "hardware revision not supported" under python 3.5. Examples run under python 2.7 with no issues.

`pi@rpi:~/ $ sudo python3 strandtest.py`
`Traceback (most recent call last):`
  `File "strandtest.py", line 97, in <module>`
    `strip.begin()`
  `File "/usr/local/lib/python3.5/dist-packages/rpi_ws281x/rpi_ws281x.py", line 130, in begin`
    `raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))`
`RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)`

furbrain, 2019-11-27:
I've also come across this problem
The issue seems to be that the .so file bundled as a wheel for Python3.5 (haven't tested other python versions) from pypi is old and doesn't have support for Pi3A. 

You can rectify it by uninstalling it, and cloning this project directly and installing it as below

Ideally we could do with uploading a new binary wheel to PyPi
```
git clone --recurse-submodules https://github.com/rpi-ws281x/rpi-ws281x-python.git
cd rpi-ws281x-python/library/
sudo python3 ./setup.py install
```

11zombie11, 2020-05-12:
Just tried @furbrain solution but gives me this error:
`/usr/bin/ld: cannot find -lws2811`

Already googled it but can't find any solution.

closed 2023-05-16 by Gadgetoid