rpi-ws281x/rpi-ws281x-python #26: Cannot Build Version 4.2.2

wallarug, 2019-09-18:
Hi there!

I am getting compiler errors for this library that is preventing me from installing from pip3.  Any chance of some assistance?

```
sudo pip3 install -e .
Obtaining file:///home/pi/rpi-ws281x-python/library
Installing collected packages: rpi-ws281x
  Running setup.py develop for rpi-ws281x
    Complete output from command /usr/bin/python3 -c "import setuptools, tokenize;__file__='/home/pi/rpi-ws281x-python/library/setup                                                                                             .py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))                                                                                             " develop --no-deps:
    running develop
    running egg_info
    writing dependency_links to rpi_ws281x.egg-info/dependency_links.txt
    writing top-level names to rpi_ws281x.egg-info/top_level.txt
    writing rpi_ws281x.egg-info/PKG-INFO
    reading manifest file 'rpi_ws281x.egg-info/SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file 'rpi_ws281x.egg-info/SOURCES.txt'
    running build_ext
    building '_rpi_ws281x' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fdebug-prefix-map=/build/python3.5-6waWnr                                                                                             /python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -Ilib/ -I/usr/inc                                                                                             lude/python3.5m -c rpi_ws281x_wrap.c -o build/temp.linux-armv7l-3.5/rpi_ws281x_wrap.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fdebug-prefix-map=/build                                                                                             /python3.5-6waWnr/python3.5-3.5.3=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/                                                                                             temp.linux-armv7l-3.5/rpi_ws281x_wrap.o -Llib-built/ -lws2811 -o build/lib.linux-armv7l-3.5/_rpi_ws281x.cpython-35m-arm-linux-gnueab                                                                                             ihf.so
    /usr/bin/ld: cannot find -lws2811
    collect2: error: ld returned 1 exit status
    error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

    ----------------------------------------
Command "/usr/bin/python3 -c "import setuptools, tokenize;__file__='/home/pi/rpi-ws281x-python/library/setup.py';f=getattr(tokenize,                                                                                              'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" develop --no-deps" fai                                                                                             led with error code 1 in /home/pi/rpi-ws281x-python/library/
```

It happens with `pip3 install rpi-ws281x` as well.

Gadgetoid, 2020-01-16:
This is now fixed in git, thanks to the efforts of @ninioperdido and should be fixed when 4.2.4 is released.

ninioperdido, 2020-01-16:
Anytime @Gadgetoid :)