rpi-ws281x/rpi-ws281x-python #32: selected GPIO not possible

Vechus, 2019-12-06:
I'm running this library on a Raspberry Compute Module 3, but I can't get it to work.
Particularly, I edited `strandtest.py`, changing the following:
```Python
LED_COUNT = 12
LED_PIN = 45
LED_CHANNEL = 1
```
Plus I changed pin 45 mode to PWM1, by issuing `gpio mode 45 ALT0`.
But when I run `python3 strandtest.py` I get the following:
```
Traceback (most recent call last):
...
... rpi_ws281x.py", line 130, in begin
raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -11 (Selected GPIO not possible)
```

However, it looks like the setup is fine. Am I missing something?

qinzheng, 2019-12-13:
I declared three strip objects  and  the same error

`
LED_1_COUNT = 512        # Number of LED pixels.
LED_1_PIN = 18          # GPIO pin connected to the pixels (must support PWM! GPIO 13 and 18 on RPi 3).
LED_1_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_1_DMA = 10          # DMA channel to use for generating signal (Between 1 and 14)
LED_1_BRIGHTNESS = 255  # Set to 0 for darkest and 255 for brightest
LED_1_INVERT = False    # True to invert the signal (when using NPN transistor level shift)
LED_1_CHANNEL = 0       # 0 or 1
LED_1_STRIP = ws.WS2811_STRIP_RGB

LED_2_COUNT = 512        # Number of LED pixels.
LED_2_PIN = 13          # GPIO pin connected to the pixels (must support PWM! GPIO 13 or 18 on RPi 3).
LED_2_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_2_DMA = 10          # DMA channel to use for generating signal (Between 1 and 14)
LED_2_BRIGHTNESS = 255  # Set to 0 for darkest and 255 for brightest
LED_2_INVERT = False    # True to invert the signal (when using NPN transistor level shift)
LED_2_CHANNEL = 1       # 0 or 1
LED_2_STRIP = ws.WS2811_STRIP_RGB

LED_3_COUNT = 512        # Number of LED pixels.
LED_3_PIN = 21          # GPIO pin connected to the pixels (must support PWM! GPIO 13 or 18 on RPi 3).
LED_3_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_3_DMA = 10          # DMA channel to use for generating signal (Between 1 and 14)
LED_3_BRIGHTNESS = 255  # Set to 0 for darkest and 255 for brightest
LED_3_INVERT = False    # True to invert the signal (when using NPN transistor level shift)
LED_3_CHANNEL = 1       # 0 or 1
LED_3_STRIP = ws.WS2811_STRIP_RGB
`

and I get the following:

`
Traceback (most recent call last):
  File "start.py", line 578, in <module>
    strip3.begin()
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 130, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -11 (Selected GPIO not possible)
`

qinzheng, 2019-12-13:
I solved the problem by changing LED3`s channel to 0

Vechus, 2019-12-13:
But GPIO 45 is PWM1, that means channel 1 no?

qinzheng, 2019-12-14:
maybe the channel  is not correspond with PWM

closed 2023-05-16 by Gadgetoid