rpi-ws281x/rpi-ws281x-python #34: WS2815 support

jandechent, 2020-01-01:
I'm new, but so far the hardward seems to be working. I have a WS2815b strip with 120 leds. I can start the example strandtest, but the color is completely off, compared to the code. First, only some LED turn on (7), then till LED 34 and never have I managed to get all light up. It is hard to describe, e.g.: colorWipe(strip, Color(255, 0, 0))  or (0,255,0) or (0,0,255)
first 19 led turn white, some flicker but led 20 or further never turns on.

I tried to pass PixelStrip some the arguments like rpi_ws281x.WS2812_STRIP, but this didn't help and there is no matching variable for a WS2815...

Does anybody have an idea?

Gadgetoid, 2020-01-16:
This library is just a wrapper for rpi_ws281x where all the magic happens - https://github.com/jgarff/rpi_ws281x

You could ask for WS2815B support there, but you'd be lucky to find someone who has WS2815B's, has some understanding of the codebase and has the time/inclination to add it. rpi_ws281x hasn't seen anything more than tweaks and fixes for over two years.

Xartrick, 2020-03-04:
I just bought a WS2815B and have no issue with `rpi_ws281x`. `strand.py` run flawlessly.

I use a logic level shifter from 3.3V to 5V between PWM and DATA signal of the strip powered by the +3.3V and +5V of the Raspberry Pi (I use the Raspberry Pi 1B).

Also, don't forget to share the GND between your external power supply and the Rasberry Pi, this caused random display on my end.

chymerix, 2020-08-10:
I'm not able to get this to work with the WS2815B. I just get random colors on the strip. Works flawlessly with the WS2812B, though - any ideas?

closed 2023-05-16 by Gadgetoid