rpi-ws281x/rpi-ws281x-python #41: Raspberry Pi 4 support (Ubuntu 18.04)

waspinator, 2020-03-23:
I'm having trouble getting things working on a Raspberry Pi 4 with Ubunut 18.04. When running the following script I get the following error:

```
#!/usr/bin/python3

import time
from rpi_ws281x import PixelStrip, Color

# LED strip configuration:
LED_COUNT = 16        # Number of LED pixels.
LED_PIN = 21          # GPIO pin connected to the pixels (18 uses PWM!).
# LED_PIN = 10        # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
LED_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA = 10          # DMA channel to use for generating signal (try 10)
LED_BRIGHTNESS = 50  # Set to 0 for darkest and 255 for brightest
LED_INVERT = False    # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53

# Define functions which animate LEDs in various ways.
def colorWipe(strip, color, wait_ms=50):
    """Wipe color across display a pixel at a time."""
    for i in range(strip.numPixels()):
        strip.setPixelColor(i, color)
        strip.show()
        time.sleep(wait_ms / 1000.0)

def set_pixels():

    strip = PixelStrip(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)
    strip.begin()
    colorWipe(strip, Color(255, 0, 0))  # Red wipe

if __name__ == '__main__':
    set_pixels() 
```

```
Traceback (most recent call last):
  File "test.py", line 31, in <module>
    set_pixels()
  File "test.py", line 27, in set_pixels
    strip.begin()
  File "/usr/local/lib/python3.6/dist-packages/rpi_ws281x/rpi_ws281x.py", line 130, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
```

I installed the library using `sudo pip3 install rpi_ws281x`.

Is the Raspberry Pi 4 running Ubuntu 18.04 supported?

Gadgetoid, 2020-03-23:
I double-checked the source package for rpi_ws281x and it has an entry for revision `c03111`. Is your last paste the end of `/proc/cpuinfo`?

It might be worth compiling from source to be sure, if memory serves me it's:

```
git clone https://github.com/rpi-ws281x/rpi-ws281x-python
cd rpi-ws281x-python
git submodule update --init
cd library
python3 setup.py build
sudo python3 setup.py install
```

waspinator, 2020-03-23:
yes, that last bit is from `/proc/cpuinfo`

I compiled from source using your instructions and have similar results to before, with a bonus segmentation fault. I ran `sudo pip3 uninstall rpi_ws281x` before installing the source version.

```
  File "/usr/local/lib/python3.6/dist-packages/rpi_ws281x-4.2.3-py3.6-linux-aarch64.egg/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
Segmentation fault (core dumped)
```

Gadgetoid, 2020-03-23:
The library lives here: https://github.com/jgarff/rpi_ws281x

And the Raspberry Pi revision you're running is here:

https://github.com/jgarff/rpi_ws281x/blob/6a720cbd42d30be28e0f5c5ff6b1c00a4588a29b/rpihw.c#L72-L78

waspinator, 2020-03-23:
maybe that's the issue. When I run `cat .gitmodules`

```
[submodule "library/lib"]
	path = library/lib
	url = http://github.com/pimoroni/rpi_ws281x
	branch = master
```

instead of the `jgarff` version. But it seems that the definition for my hardware revision is in that repo too. There is a difference on line 419

https://github.com/pimoroni/rpi_ws281x/blob/1e900b2c1af5d9365e275ad4f77ee01270851483/rpihw.c#L419
https://github.com/jgarff/rpi_ws281x/blob/6a720cbd42d30be28e0f5c5ff6b1c00a4588a29b/rpihw.c#L419

running with 419 from jgarff seems to work but when using 419 from pimoroni I get a `Segmentation fault (core dumped)`

```
#include <stdint.h>
#include <stdio.h>
#include <string.h>
#include <stdlib.h>
#include <errno.h>
#include <byteswap.h>

#define LINE_WIDTH_MAX                           80
#define HW_VER_STRING                            "Revision"

#define PERIPH_BASE_RPI                          0x20000000
#define PERIPH_BASE_RPI2                         0x3f000000
#define PERIPH_BASE_RPI4                         0xfe000000

#define VIDEOCORE_BASE_RPI                       0x40000000
#define VIDEOCORE_BASE_RPI2                      0xc0000000

int main() {

typedef struct {
    uint32_t type;
#define RPI_HWVER_TYPE_UNKNOWN                   0
#define RPI_HWVER_TYPE_PI1                       1
#define RPI_HWVER_TYPE_PI2                       2
#define RPI_HWVER_TYPE_PI4                       3
    uint32_t hwver;
    uint32_t periph_base;
    uint32_t videocore_base;
    char *desc;
} rpi_hw_t;
   
static const rpi_hw_t rpi_hw_info[] = {
    //
    // Raspberry Pi 4
    //
    {
        .hwver = 0xA03111,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 1GB v1.1"
    },
    {
        .hwver = 0xB03111,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 2GB v.1.1"
    },
    {
        .hwver = 0xC03111,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 4GB v1.1"
    }

};

   const rpi_hw_t *result = NULL;
   uint32_t rev;
   unsigned i;

   #ifdef __aarch64__
   printf("__aarch64__\n");
   #else
      printf("not\n");
   #endif

   FILE *f = fopen("/proc/device-tree/system/linux,revision", "r");
   size_t read = fread(&rev, 1, sizeof(uint32_t), f);

    if (read != sizeof(uint32_t)) {
        printf("bad size\n");
        goto done;
    }
        
    #if __BYTE_ORDER__ == __ORDER_LITTLE_ENDIAN__
        printf("swap\n");
        rev = bswap_32(rev);  // linux,revision appears to be in big endian
    #endif

    for (i = 0; i < (sizeof(rpi_hw_info) / sizeof(rpi_hw_info[0])); i++)
    {
        uint32_t hwver = rpi_hw_info[i].hwver;
        if (rev == hwver)
        {
            result = &rpi_hw_info[i];

            goto done;
        }
    }

done:
   fclose(f);
   printf(result->desc);
   printf("\n");
   return 0;
}
```

output when using jgarff
```
__aarch64__
swap
Pi 4 Model B - 4GB v1.1
```

dheera, 2020-03-29:
Confirmed that replacing the library with rpi_ws281x 6a720cbd42d30be28e0f5c5ff6b1c00a4588a29b works.

closed 2023-05-16 by Gadgetoid