rpi-ws281x/rpi-ws281x-python #46: Segfault on 4.2.4

jonathanhacker, 2020-06-15:
With version 4.2.4 python scripts segfault after creating an Adafruit_NeoPixel instance.

```
Python 3.7.5 (default, Nov  7 2019, 10:50:52)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rpi_ws281x
>>> rpi_ws281x.Adafruit_NeoPixel(0, 0)
<rpi_ws281x.rpi_ws281x.Adafruit_NeoPixel object at 0x7f2fff132110>
>>> exit()
zsh: segmentation fault (core dumped)  python
```

Same behavior in the python:3 container from dockerhub (python version 3.8.2).

storca, 2020-06-28:
Try using PixelStrip object (in examples/strandtest.py) instead of Adafruit_NeoPixel.
If that doesn't work for you, check #47

jonathanhacker, 2020-06-29:
PixelStrip gives me the same behavior. Not very surprising, as they are both pretty much [the same class](https://github.com/rpi-ws281x/rpi-ws281x-python/blob/master/library/rpi_ws281x/rpi_ws281x.py#L183).
```
Python 3.7.5 (default, Nov  7 2019, 10:50:52)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rpi_ws281x
>>> rpi_ws281x.PixelStrip(0,0)
<rpi_ws281x.rpi_ws281x.PixelStrip object at 0x7f2dfa8012d0>
>>> exit()
```

Building the library myself did not fix the issue either.
Currently I am simply using 4.2.3.

storca, 2020-06-30:
Try supplying more parameters to the object's construcor then?
Dumb suggestion but may work

jonathanhacker, 2020-07-01:
```
import rpi_ws281x

LED_COUNT = 16  # Number of LED pixels.
LED_PIN = 10  # GPIO pin used. 10: SPI, 18: PWM (used for sound)
LED_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA = 10  # DMA channel to use for generating signal (try 10)
LED_BRIGHTNESS = 255  # Set to 0 for darkest and 255 for brightest
LED_INVERT = False  # True inverts the signal (when using NPN level shift)
LED_CHANNEL = 0  # set to '1' for GPIOs 13, 19, 41, 45 or 53

rpi_ws281x.Adafruit_NeoPixel(
    LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL,
)
```
This is how I'm using it. Still same behavior on 4.2.4.

storca, 2020-07-03:
Weird... Maybe specifying your hardware may help?

jrknight, 2020-09-24:
I'm having the exact same issue on a raspberry pi zero.

`Fatal Python error: Segmentation fault

Current thread 0xb6f04010 <python> (most recent call first):
  File "/usr/local/lib/python2.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 54 in __setitem__
  File "/usr/local/lib/python2.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 143 in setPixelColor
  File "/root/rpi_ws281x/python/lightstest/lights.py", line 40 in rainbowCycle
  File "lightstest/server.py", line 13 in <module>
Segmentation fault`

jrknight, 2020-09-24:
Following up on this, I've examined @jonathanhacker 's code and it looks like I was missing the following method call on my strip:

Gadgetoid, 2023-05-16:
Closed as stale and (hopefully?) solved.

closed 2023-05-16 by Gadgetoid