rpi-ws281x/rpi-ws281x-python #47: [RPi Zero] RuntimeError: ws2811_init failed with code -6 (Unable to map registers into userspace)

storca, 2020-06-27:
Hello
I have a script that does the same as the example, in the example below, I did sudo python3

```python
Python 3.7.3 (default, Dec 20 2019, 18:57:59) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import rpi_ws281x as r
>>> strip = r.Adafruit_NeoPixel(12,18,800000,255)
>>> strip.begin()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -6 (Unable to map registers into userspace)
>>> exit()
Segmentation fault
```

Thanks for any help

storca, 2020-06-28:
Fixed the issue by using the object PixelStrip instead of Adafruit_NeoPixel as seen in the example strandtest.py

If that doesn't work for you, try building the library yourself (python3 here)

```
git clone https://github.com/rpi-ws281x/rpi-ws281x-python.git --recurse-submodules
cd rpi-ws281x-python/library
umask 022
sudo pip3 install .
```

closed 2020-06-28 by storca

alexwhittemore, 2021-07-20:
FYI for anyone in my shoes who ends up here: If you see

`RuntimeError: ws2811_init failed with code -5 (mmap() failed)`

it's probably because you need to be running your program as root/with `sudo`.