rpi-ws281x/rpi-ws281x-python #48: Detected a memory leak of type 'ws2811_t *', no destructor found.

nicofirst1, 2020-07-02:
As mentioned [here](https://github.com/jgarff/rpi_ws281x/issues/263#issuecomment-652890701) I am trying to to run a custom script where I initiate the class as:

```
from rpi_ws281x import PixelStrip
np= PixelStrip(pixels,PIN) 
np.begin()
```

I am working on Raspberry Pi 4 Model B Rev 1.2 with Python 3.7.3 and running the script with sudo.
The error is the following:
```
Traceback (most recent call last):
  File "src/raspberrypi/main.py", line 23, in <module>
    app = pat(handler=PiHandler, rate=rate, pixels=64, color=color)
  File "/home/pi/Repos/ledypie/src/patterns/chasing.py", line 13, in __init__
    super().__init__(**kwargs)
  File "/home/pi/Repos/ledypie/src/patterns/default.py", line 19, in __init__
    self.handler=handler(rate)
  File "/home/pi/Repos/ledypie/src/raspberrypi/pi_handler.py", line 11, in __init__
    self.np= PixelStrip(pixels,PIN)
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py", line 95, in __init__
    ws.ws2811_channel_t_count_set(self._channel, num)
TypeError: in method 'ws2811_channel_t_count_set', argument 2 of type 'int'
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.
```
I installed the library with `sudo pip3 install rpi_ws281x` which yielded no errors.

nicofirst1, 2020-07-03:
I finally fixed the issue.
I was passing the `pixels` value to the `PixelStrip(pixels,PIN)` as a float instead of an int.

closed 2020-07-03 by nicofirst1

danpe, 2022-08-05:
For me it happens straight from the example  https://github.com/rpi-ws281x/rpi-ws281x-python/blob/3f7f031b8a87f584fcef72225b7cfb1cff682316/examples/lowlevel.py#L65

WbrJr, 2023-08-27:
@danpe same here. Do you remember by any chance how you fixed it?