rpi-ws281x/rpi-ws281x-python #50: Is there a way to control each pixel's brightness? I know that `PixelStrip` class has a method `setBrightness()`, but that modifies the brightness of the entire strip.

jwnicholas99, 2020-08-07:


Gadgetoid, 2020-08-07:
WS2812s don't have a global brightness function like APA102s, so any individual pixel brightness would just be a multiplier on the R, G and B colour channels. It's theoretically possible to implement, but scaling brightness in this way usually has some significant repurcussions to colour accuracy so it might not be sensible to make it a feature.

Basically setting the brightness of an individual pixel would be no different to `setPixelColor(x, r * brightness, g * brightness, b * brigthtness)`

jwnicholas99, 2020-08-07:
I see, I get what you mean. I'm thinking about creating more complex animations for the LED strip and hence wanted to modify the brightness of each pixel individually. Thanks!

closed 2020-08-07 by jwnicholas99