rpi-ws281x/rpi-ws281x-python #52: Is there a way to cleanup the resources used by Adafruit_NeoPixel?

Looking4OffSwitch, 2020-09-20:
Is there no way to reset a strip so that it can be reused without having to restart my Python script? I see that the `Adafruit_NeoPixel` has a `_cleanup()` method but doing the following results in unpredictable behavior (i.e. the LEDs don't function as expected):

```
clock = LEDClock(led_segments=32, leds_per_segment=9, pin=18, freq=800000, dma=10,
                 brightness=200, invert=False, channel=0, strip_type=ws.WS2812_STRIP)
clock.strip._cleanup()

clock2 = LEDClock(led_segments=32, leds_per_segment=9, pin=18, freq=800000, dma=10,
                  brightness=200, invert=False, channel=0, strip_type=ws.WS2812_STRIP)
```

In my project, the strip can be used for multiple tasks: a clock, a countdown timer, scrolling text, etc. It would be nice if I could "deallocate" the underlying resources so that each task (Python class) could create and manage it's own `Adafruit_NeoPixel` instance.

Gadgetoid, 2020-09-20:
Until the memory leak(s) in the underlying rpi_w281x library is found and fixed then you neeed to create and manage just a single instance, or you'll run headlong into it. There's nothing the Python wrapper can do about this.

See- https://github.com/jgarff/rpi_ws281x/issues/390

Looking4OffSwitch, 2020-09-21:
Thank you so much for your response @Gadgetoid!

closed 2020-09-21 by Looking4OffSwitch