rpi-ws281x/rpi-ws281x-python #53: RuntimeError: wx2811_init failed with code -5 (mmap() failed)

charliesneath, 2020-09-25:
I’m working on getting a new strip up and running and am testing it out with the included `strandtest.py` script. I’m getting the below error after installing and running the following command. This looks like a permission error, but I am using `sudo`, so any additional guidance is appreciated.

`sudo python3.7 strandtest.py --clear`

(Note that the error message is transcribed since I wasn’t able to copy and paste from my RPi).

```
Can't open /dev/mem: Permission denied
Traceback (most recent call last):
    File “/home/pi/rpi-ws281x-python/examples/strandtest.py”, line 97, in <module>
        strip.begin()
    File “/usr/local/lib/python3.7/dist-packages/rpi_ws281x/rpi_ws281x.py”, line 131, in begin
        raise RuntimeError(‘ws2811_init failed with code {0} ({1}}).format(resp, str_resp))
RuntimeError: wx2811_init failed with code -5 (mmap() failed)
```

tv42, 2020-11-13:
Run it as root. It needs very low-level hardware access.

charliesneath, 2020-11-19:
Closing because I'm not running into this anymore. I was running as root but have made other changes and I'm not sure how this was resolved.

closed 2020-11-19 by charliesneath