rpi-ws281x/rpi-ws281x-python #54: Flashing leds on moving rainbow

Zoraiyo, 2020-10-31:
I have a SK6812 (description similar to WS2812B) led strip with 600 leds over 10m. Beside the RGB led it also has a warm white led. It is connected to a RPi 3B. The leds are not in one piece, but distributed over 20m in 3 large areas.

Displaying single colors or a standing rainbow works without problems. Till now I had used a modified version of the wheel function (with changable max positions/leds). But when I let the rainbow move, the RGB or/and the white LED flashes randomly in full brightness. So I adopted the code from https://github.com/rpi-ws281x/rpi-ws281x-python/issues/40#issuecomment-713011986, but the flashing still occurs. I tried different update rates (less flashing with1s, more with 1/60s). The color of each pixel is set with the correct value.

Is it possible that my signal cable is too long, so that the fast update of the strip will result in faulty transmissions? 
Or do I need to adjust my update rate to the amount of leds?

Thanks a lot for the help

stsdc, 2021-02-05:
Hello, @Zoraiyo! I'm also experiencing this issue. Did you manage to resolve your issue?

Zoraiyo, 2021-05-02:
I have found a solution: 
I have added a 30 Ohm resistor in serial between the PWM Pin and the first led. Now I can update the stripe with 60 Hz without flickering.

closed 2021-05-02 by Zoraiyo