rpi-ws281x/rpi-ws281x-python #55: sudo needed to run

zeesh1979, 2020-11-04:
Hi,
I am using RPI3 and added pi to gpio and kmem groups but still i get permission denied error when running the script. running with sudo works

Can't open /dev/mem: Permission denied
Traceback (most recent call last):
  File "rgb1.py", line 84, in <module>
    strip.begin()
  File "/usr/local/lib/python3.5/dist-packages/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -5 (mmap() failed)
Segmentation fault

Zoraiyo, 2020-11-07:
Hi zeesh1979, 

I read somewhere in the past, that "sudo" is required to access the gpio/PWM. Sadly I don't now where I read that. 
I run my script with sudo too. 

Best regards

Rapti, 2021-01-22:
I'm having a similar problem although the error message is different. Using a Raspberry Pi 4B with Arch Linux ARM.

```
$ python monitor_server.py 
Failed to create mailbox device
: Operation not permitted
Traceback (most recent call last):
  File "/home/openhab/LED/monitor_server.py", line 160, in <module>
    strip.begin()
  File "/usr/lib/python3.9/site-packages/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -9 (Failed to create mailbox device)
Speicherzugriffsfehler (Speicherabzug geschrieben)
```

Running the script as root works, but I'd very much like to avoid that for obvious reasons. Is there any group I can add the user to to fix this? Being a member of `gpio` does the trick with other libraries using GPIO pins, but it doesn't seem to be enough in this case. I couldn't find anything about this on the internet.

MainzerKaiser, 2021-06-21:
Did anybody find a solution for that? I get a similar error:
`Can't open /dev/mem: Keine Berechtigung
Traceback (most recent call last):
  File "/home/pi/PycharmProjects/LEDStrip/main.py", line 25, in LEDStartfunct
    self.LEDStart()
  File "/home/pi/PycharmProjects/LEDStrip/LEDStart.py", line 101, in LEDStart
    strip.begin()
  File "/home/pi/PycharmProjects/LEDStrip/venv/lib/python3.7/site-packages/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -5 (mmap() failed)`

I simply would like to initiate the LEDstart function from another program wihout executing a command string having sudo in it.

ConnorDY, 2021-12-29:
I'm having the same problem that @Rapti described. I'm also using this with a Raspberry Pi 4B running Arch Linux ARM.

sdparkes66, 2022-01-02:
Seeing same thing here PiZero 2 w

Script runs with SUDO.

```
RuntimeError: ws2811_init failed with code -9 (Failed to create mailbox device)
swig/python detected a memory leak of type 'ws2811_t *', no destructor found.
(venv) ******@PiZero1:~/Python/LEDring $ cat /proc/cpuinfo | grep Revision

```

```
cat /proc/cpuinfo | grep Revision
Revision        : 902120
```

zeesh1979, 2022-01-02:
no admin responding, i am closing it

closed 2022-01-02 by zeesh1979

Gadgetoid, 2022-01-04:
This library is just a Python wrapper around https://github.com/jgarff/rpi_ws281x/

`sudo` is required.

sdparkes66, 2022-01-04:
> Good to know and now understand, thanks