rpi-ws281x/rpi-ws281x-python #56: Raspberry Pi 4 Model B Rev 1.4 not working with 2.4.2

RussellTaylor83, 2020-12-26:
Hardware        : BCM2711
Revision        : c03114
Serial          : 100000007781f4f7
Model           : Raspberry Pi 4 Model B Rev 1.4

```
File "/home/pi/git/Freenove_Robot_Dog_Kit_for_Raspberry_Pi/Code/Server/Led.py", line 22, in __init__
    self.strip.begin()
  File "/usr/local/lib/python3.7/dist-packages/rpi_ws281x-4.2.4-py3.7-linux-armv7l.egg/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
Segmentation fault
```

We'll need to add this to rpihw.c

```
{
        .hwver = 0xc03114,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Pi 4 Model B - 4GB v1.4"
    },
```

I have a new 8GB version at home I will also look out the hwversion. I'll try to get a PR up asap, then maybe it's worth doing a release. The only reason I say this is I'm here because my robot dog Christmas present isn't working and I'd hate other people who are less software savvy to be in the same boat.

Thank you for all the work on this library.

RussellTaylor83, 2020-12-26:
I've just realised this library relies on a submodule from pimoroni

https://github.com/pimoroni/rpi_ws281x

I'll fix it up there and report back.

cp2004, 2021-01-01:
Looks like there's an open PR here https://github.com/jgarff/rpi_ws281x/pull/438, now I just need to figure out how I can build this myself, and package the python dependency so I can send it to users with the same issue. I'm maintaining [OctoPrint WS281x LED Status](https://github.com/cp2004/OctoPrint-WS281x_LED_Status). Need to also work out the versioning here, since there is V4.2.5 published to PyPi but the source says 4.2.4 & is tagged as such.

RussellTaylor83, 2021-01-01:
Sorry, I forgot I said I would do this. I must admit, the versioning and notes confused me slightly. Good spot on that PR, I'll check it over later but it looks like it has both my boards.

So for anyone reading, this is how I fixed it for my use case:

sudo pip3 uninstall rpi_ws281x
git clone --recurse-submodules https://github.com/rpi-ws281x/rpi-ws281x-python
vim rpi-ws281x-python/library/lib/rpihw.c
make the changes needed
cd rpi-ws281x-python/library/
sudo python3 setup.py install

How you do that in the context of github / your library I'm not quite sure. I'll try to find some time tonight to look at it.

cp2004, 2021-01-01:
[rpi_ws281x-4.2.4.zip](https://github.com/rpi-ws281x/rpi-ws281x-python/files/5759210/rpi_ws281x-4.2.4.zip)

I made the edits you suggested, and built it into a sdist zip. Can be installed using pip, this was easiest for me since OctoPrint's plugin manager allows for uploading these to install using pip. If anyone else runs into this issue, download the file above and run `python -m pip install rpi_ws281x-4.2.4.zip` and it should work. Version number is still off as I didn't change that, so it would update nicely when I pull in the dependencies inside the actual plugin.

Thanks for the pointers!

Gadgetoid, 2021-03-09:
Try 4.2.6 from - https://test.pypi.org/project/rpi-ws281x/

Only have a test source package up at the moment because my usual build Pi is in a box somewhere under a pile of other boxes.

DrWatson, 2021-03-09:
Hmm:

`pi@HyperBian:/ $ pip install -i https://test.pypi.org/project/rpi-ws281x rpi-ws281x
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://test.pypi.org/project/rpi-ws281x
Requirement already satisfied: rpi-ws281x in /home/pi/.local/lib/python3.7/site-packages (4.2.5)
`

DrWatson, 2021-03-09:
Had to uninstall it first:
`Successfully built rpi-ws281x
Installing collected packages: rpi-ws281x
Successfully installed rpi-ws281x-4.2.6
`
Restarted the pi but still got this in the logs:
` 2021-03-09T16:39:24.987Z [hyperiond LEDDEVICE] (ERROR) Device disabled, device 'ws281x' signals error: 'Failed to open. Error message: Hardware revision is not supported'`

cp2004, 2021-03-09:
> Try 4.2.6 from - https://test.pypi.org/project/rpi-ws281x/
> 
> Only have a test source package up at the moment because my usual build Pi is in a box somewhere under a pile of other boxes.

@Gadgetoid if you want something built & tested I can help there. Obviously I can't upload it to PyPi, and if the lack of a wheel is blocking the full release (on actual PyPi) remember that PiWheels exists and will likely build the wheel from the sdist anyway, so there will be no noticeable difference for users - https://www.piwheels.org/project/rpi-ws281x/

cp2004, 2021-03-09:
> @cp2004 thank you- I think I can safely rely on PiWheels but the main blocker was being unable to run this build and make sure I haven't done something stupid. If you can successfully run the test package I'm happy to kick it to release.

Tested on my Pi 4s, both rev 1.1, 2GB and 4GB under my plugin with WS2811b, all seems to be going just fine.