rpi-ws281x/rpi-ws281x-python #57: RGB LED HAT for Raspberry Pi

RobVor, 2021-01-09:
Hi All,

Got an RGB LED HAT for Raspberry Pi based on the WS2812B.

https://www.waveshare.com/product/raspberry-pi/hats/led-buttons/rgb-led-hat-b.htm

https://www.waveshare.com/wiki/RGB_LED_HAT_(B)

The Wiki states the use of the rpi-ws281x as the LED 8 x 8 grid driver.

I have installed and reinstalled this now on anything from a PiZero to a Pi4 Rev 1.1 and can't get it to work.

After hashing through all of the errors, I can finally get the code to run but I am not getting any response on the HAT. No LED's are lighting up and nothing is happening.

Rather frustrating.

Has anyone else tried this or similar HAT? 
Any suggestions on how to get it to work?

Gadgetoid, 2021-01-11:
I think this HAT is virtually identical to our original 8x8 Unicorn HAT which is what I originally wrote this port of rpi_ws281x to support. It definitely *used* to work, and I haven't seen any complaints from customers that suggest it's since broken.

I *guess* you could try the Unicorn HAT driver - https://github.com/pimoroni/unicorn-hat

RobVor, 2021-01-11:
Thanks. Will have a go with that now.

closed 2023-05-16 by Gadgetoid