rpi-ws281x/rpi-ws281x-python #60: Add Numpy support

stsdc, 2021-02-13:
Will numpy speed up an execution of `setPixelColorRGB()`?

I'm currently creating numpy array based on the audio signal, so latency is important. Unfortunately it is impossible to pass numpy array to a `setPixelColorRGB()` since it is not in the standard Python `int` type. I'm wondering if it is a bottleneck or is a Python as a whole?

cp2004, 2021-04-24:
> Will numpy speed up an execution of setPixelColorRGB()?

Probably not, unless it can do some magic to make it much faster. The underlying code for this lib is written in C.   Adding huge dependencies such as Numpy would complicate installation.

There's very little Python code around, it mostly goes straight to the C side. Could you share some data on what you are seeing as a bottleneck? I've been able to push the LEDs quite fast in some tests.

Gadgetoid, 2023-05-16:
Closed as stale.

closed 2023-05-16 by Gadgetoid