rpi-ws281x/rpi-ws281x-python #64: Need script to run on boot and in the background

bobster316, 2021-03-22:
I am trying to get one of the example scripts to run when the raspberry pi 4 boots up and continue to run in the background.

I have tried adding the following to the crontab -
@reboot python3 /home/pi/rpi_ws281x/python/examples/strandtest.py &
@reboot sudo python3 /home/pi/rpi_ws281x/python/examples/strandtest.py &

When the pi boots is the led's come on but do not change.

The script works fine if run as a regular command from the prompt

lbt, 2021-09-15:
Using crontab is a real bodge and although some things may work it's a terrible approach and isn't worth learning.

Instead try learning about systemd which is the "proper" modern way to do exactly what you want.

I searched and found this tutorial:
https://tecadmin.net/setup-autorun-python-script-using-systemd/

To do it properly you should use the path /etc/systemd/system and not /lib/systemd/system as mentioned in that tutorial

Gadgetoid, 2023-05-16:
Closed as stale. Running on boot is outside the scope of the library, and down to individual implementations.

closed 2023-05-16 by Gadgetoid