rpi-ws281x/rpi-ws281x-python #70: Less than 6 LEDs results in flickering and wrong colors of LEDs

cp2004, 2021-06-27:
This was reported to me as a downstream maintainer, but I have also confirmed it myself. Report (https://github.com/cp2004/OctoPrint-WS281x_LED_Status/issues/132).

**The problem**
Using less than 6 LEDs in the strip initializer creates incorrect colours. It also, for example when trying to use the rainbow effect (from strandtest) flickers lots.

A solid colour of red when I set 5 LEDs, actually creates a gradient from kind of pinkish colour to blue on the 5th LED.
Setting to 6 LEDs sets full red, as expected.

This doesn't seem to be the case when using the C library, there's no 1:1 strandtest.py-type script that I can find to actually work. The matrix style one seems to work the same across all configured lengths of LEDs.

**Steps to reproduce**
1. LEDs connected to pin 10, using SPI, properly configured and normally working
2. Install this library, and use the strandtest.py script with `LED_COUNT = 5` or less
3. Run script, observe lots of incorrect colours.

I can reproduce this, and it was reported by another user.
This doesn't seem to be a problem with running the test straight from the C library, without the Python bindings. I am currently unable to test if it also occurs using PWM/GPIO 18.

**System Info**
* Raspberry Pi 4 Model B Rev 1.1
* OctoPi 0.18 - Image based on Raspberry Pi OS from December 2020, up to date
* Python 3.7.3 & 2.7.16

I have been trying to debug this, but have no idea where to start. C code confuses me. If there is anything that could be looked at properly with a debugger, I can try and do it.

Gadgetoid, 2023-05-16:
Does this still happen as of v5.0.0?

cp2004, 2023-05-16:
I will add it to my list of things to test, but it will not be for a couple of weeks I think. I will try and let you know as soon as I can.