rpi-ws281x/rpi-ws281x-python #71: after update raspi version to 5.10.17, rpi_ws281x cannot control ledstrip

dfybc, 2021-07-29:
after update raspi version to 5.10.17, rpi_ws281x cannot control ledstrip
any one know a solution?

lbt, 2021-09-15:
Could it be linked to this https://github.com/jgarff/rpi_ws281x/issues/447

dfybc, 2021-11-12:
> Could it be linked to this [jgarff/rpi_ws281x#447](https://github.com/jgarff/rpi_ws281x/issues/447)

I think it is not  a same issue.

my raspios version is 2021-05-07 buster, hardware is raspi 3B, 
i control ledstrip with PWM on pin18, 
but no led will light. 
any body know why?

ps: I test on lastest bullseye version, it does not work  either.

x3l51, 2022-03-11:
Can you share the wiring and settings on the rPi you did? I've run into the same problem and couldn't solve it even by changing rPi versions.

closed 2023-05-16 by Gadgetoid