rpi-ws281x/rpi-ws281x-python #86: RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)

jonhardman, 2022-06-30:
Hello,

I am struggling to resolve a 'Hardware revision is not supported' error message.
I have installed rpi_ws281x python on a raspberry PI compute module 4 (revision c03141).

The install seems to work successfully using the following command:

sudo pip install rpi_ws281x

However when it comes to running one of the standard examples I get the following error:

RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)

Python version: 3.9.2
Python3 version: 3.9.2

Any help / tips will be greatly received :-)

Jon

NeusAap, 2022-06-30:
What CM4 exactly? According to rpihw.c, these are the currently supported CM4´s:

    // Compute Module 4
    //
    {
        .hwver = 0xa03140,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Compute Module 4 v1.0 eMMC"
    },
    {
        .hwver = 0xb03140,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Compute Module 4 v1.0 Lite"
    },
    {
        .hwver = 0xc03140,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Compute Module 4 v1.0 WiFi"
    },
    {
        .hwver = 0xd03140,
        .type = RPI_HWVER_TYPE_PI4,
        .periph_base = PERIPH_BASE_RPI4,
        .videocore_base = VIDEOCORE_BASE_RPI2,
        .desc = "Compute Module 4 v1.0 WiFi 8GB"
    },

I am currently using this with a no Wifi, 4gb emmc and 4gb ram version of the CM4

marcuszerep, 2022-07-11:
See https://github.com/jgarff/rpi_ws281x/issues/483 for more details. 

Newer Raspberry Pis have different configuration values not added into that file. When you buy a new Raspberry Pi and it has this new configuration, it will throw this error.

dleidert, 2023-03-07:
Any chance to get an updated version of the Python module? [jgarff/rpi_ws281x](https://github.com/jgarff/rpi_ws281x) just released a new version, including support for multiple models.

Gadgetoid, 2023-03-07:
It's on my list. I did the `rpi_ws281x` release :grimacing:

centrek, 2023-05-05:
> It's on my list. I did the `rpi_ws281x` release 😬

Hi all, is this update still on the pipeline? :)

dleidert, 2023-05-06:
@Gadgetoid We would really appreciate an update here because we are stuck with the jgarff/rpi_ws281x#483 for some time now, and it starts blocking things.

Gadgetoid, 2023-05-16:
Is this fixed with v5.0.0?

Gadgetoid, 2024-03-28:
Note- this will not fix Pi 5.

See: https://github.com/jgarff/rpi_ws281x/issues/528

closed 2024-03-28 by Gadgetoid