rpi-ws281x/rpi-ws281x-python #90: Setting pixel colors at once

Viicos, 2023-02-27:
As of today, the only possible way (at least to my knowledge) to set multiple pixel colors is by using a for loop:

```python
for i in range(start, stop):
    strip.setPixelColor(i, color)
```

What `setPixelColor` is actually doing is:

```python
self._led_data[n] = color
```

I see that `_LED_Data` supports slicing:

https://github.com/rpi-ws281x/rpi-ws281x-python/blob/3f7f031b8a87f584fcef72225b7cfb1cff682316/library/rpi_ws281x/rpi_ws281x.py#L47-L51

But it only works this way:

```python
self._led_data[start:end] = [color for _ in range(n)]
```

I think it would be better to have slicing working the intended way, e.g. `self._led_data[start:end] = color`, and have a new `setPixelsColor` method to do so, and the current behaviour could be reproduced with a new method as well, like `setPixelsColorMap` or something.

@Gadgetoid if you are busy, I'll be happy to make a PR if you think this is a good idea.

Gadgetoid, 2023-02-27:
Almost feels like PixelStrip should just subclass `_LED_Data` which would solve this pretty handily?

Gadgetoid, 2023-02-27:
Exactly, I think that makes sense?

closed 2023-04-25 by Gadgetoid