rpi-ws281x/rpi-ws281x-python #95: Where can I find the API for the rpi-ws281x-python library?

xgqfrms, 2023-05-11:
# Where can I find the API for the rpi-ws281x-python library?

xgqfrms, 2023-05-11:
```py
#!/usr/bin/env python3
# coding: utf8

from time import sleep
from rpi_ws281x import PixelStrip, Color

# LED strip configuration:
LED_COUNT = 30        # Number of LED pixels.
LED_PIN = 18          # GPIO pin connected to the pixels (18 uses PWM!).
# LED_PIN = 10        # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
LED_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA = 10          # DMA channel to use for generating signal (try 10)
LED_BRIGHTNESS = 20   # Set to 0 for darkest and 255 for brightest
LED_INVERT = True     # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53

# Create NeoPixel object with appropriate configuration.
strip = PixelStrip(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)

# Intialize the library (must be called once before other functions).
strip.begin()
print("Test ✅")

# clear
for i in range(strip.numPixels()):
    strip.setPixelColor(i, Color(0, 0, 0))
strip.show()
sleep(1)

# Switch all pixels on
for i in range(strip.numPixels()):
    # here not work as expect ❌
    strip.setPixelColor(i, Color(255, 0, 0))
strip.show()

print("Test finished 🎉")
exit()

```

Gadgetoid, 2023-05-16:
You pretty much need to read the source. I don't have much time to maintain or support this library, and driving ws281x pixels from the Pi has always been a crapshoot.

Solid white is a really bad colour to use for testing, it's likely that your initial code doesn't work properly and the result just happens to correlate with what you're trying to accomplish. Use a logical test pattern like repeating Red/Green/Blue.

Also `LED_INVERT` should *probably* be `False` in your case, I don't see any inline transistor in your setup.

xgqfrms, 2023-06-02:
## After change to `LED_INVERT = False`

```py
#!/usr/bin/env python3
# coding: utf8

from time import sleep
from rpi_ws281x import PixelStrip, Color

# LED strip configuration:
LED_COUNT = 60        # Number of LED pixels.
LED_PIN = 18          # GPIO pin connected to the pixels (18 uses PWM!).
# LED_PIN = 10        # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
LED_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA = 10          # DMA channel to use for generating signal (try 10)
# color channel GRB ❓
LED_BRIGHTNESS = 50   # Set to 0 for darkest and 255 for brightest
LED_INVERT = False     # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53

# Create NeoPixel object with appropriate configuration.
strip = PixelStrip(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)

# Initialize the library (must be called once before other functions).
strip.begin()
print("Test beginning ✅")

def clear_buffer():
  for i in range(LED_COUNT):
    strip.setPixelColor(i, Color(0, 0, 0))
  strip.show()
  sleep(1)

def test():
  print("strip.numPixels() =", strip.numPixels())
  for i in range(strip.numPixels()):
    # white ✅
    # strip.setPixelColor(i, Color(255, 255, 255))
    # red ❌
    strip.setPixelColor(i, Color(255, 0, 0))
  strip.show()
  sleep(1)

try:
  test()
except KeyboardInterrupt:
  print('Ctrl + C exit ✅')
except RuntimeError as error:
  print("error =", error, error.args[0])
  pass
except Exception as error:
  print("exception =", error)
  raise error
finally:
  print("after three seconds, auto clear buffer! 👻")
  sleep(3.0)
  clear_buffer()
  print("Test finished 🎉")

exit()

```

xgqfrms, 2023-06-02:
# API

[image]

https://github.com/rpi-ws281x/rpi-ws281x-python/blob/master/library/rpi_ws281x/rpi_ws281x.py

closed 2023-06-02 by xgqfrms

xgqfrms, 2023-06-12:
# solution ✅

> Raspberry Pi 4B 8GB + official Power Supply (5V/3A)

Raspberry Pi `15W` USB-C Power Supply

https://www.raspberrypi.com/products/type-c-power-supply/

xgqfrms, 2023-06-12:
[image]

4B ✅

```sh
$ vcgencmd get_throttled
throttled=0x0
```

3B ❌

```sh
$ vcgencmd get_throttled
throttled=0x50000
```

MasudRana72, 2023-06-19:
https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/issues/151#:~:text=So%2C%20RPI%203B%20is%20the%20problem%3F%0ADo%20you%20fix%20the%20issue%20for%203B%3F

xgqfrms, 2023-06-19:
@MasudRana72  You can try it with the following hardware.

> Raspberry Pi `3B` + `12.5W ` official power supply ❓(I don't have this one power supply for now.)

```sh
$ vcgencmd get_throttled
```

xgqfrms, 2023-06-19:
> [adafruit/Adafruit_CircuitPython_NeoPixel#151](https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/issues/151)

see

xgqfrms, 2023-06-20:
# final solution ✅

https://www.youtube.com/watch?v=4fF5joeQ2dY

```sh
$ vcgencmd get_throttled
throttled=0x0

```

> disable audio `snd_bcm2835` ✅

```sh
$ sudo vim /boot/config.txt

$ sudo cat /boot/config.txt

# Enable audio (loads snd_bcm2835)
# ✅ fix: https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/issues/151
# dtparam=audio=on
```

```sh
$ sudo reboot

$ sudo ./neopixel-strip-rgb.py
# OR
$ sudo ./pixel-strip-rgb.py
```

```sh
$ cat ./neopixel-strip-rgb.py
```

```py
#!/usr/bin/env python3
# coding: utf8

import board
import neopixel
from time import sleep

# 60 LEDs
pixels = neopixel.NeoPixel(board.D18, 60)

def red_led_strip():
  for x in range(0, 60):
    pixels[x] = (255, 0, 0)
    sleep(0.1)
def green_led_strip():
  for x in range(0, 60):
    pixels[x] = (0, 255, 0)
    sleep(0.1)
def blue_led_strip():
  for x in range(0, 60):
    pixels[x] = (0, 0, 255)
    sleep(0.1)

def clear_buffer():
  for x in range(0, 60):
    pixels[x] = (0, 0, 0)

try:
  red_led_strip()
  sleep(1.0)
  green_led_strip()
  sleep(1.0)
  blue_led_strip()
  sleep(1.0)
except KeyboardInterrupt:
  print('Ctrl + C exit ✅')
except RuntimeError as error:
  print("error =", error, error.args[0])
  pass
except Exception as error:
  print("exception =", error)
  raise error
finally:
  print("after three seconds, auto clear buffer! 👻")
  sleep(3.0)
  # cleanup
  clear_buffer()
  print('clear 🚀')

"""
$ chmod +x ./neopixel-strip-rgb.
# ✅
$ sudo ./neopixel-strip-rgb.
"""
```

```sh
$ cat pixel-strip-rgb.py
```

```py
#!/usr/bin/env python3
# coding: utf8

from time import sleep
from rpi_ws281x import PixelStrip, Color

# LED strip configuration:
LED_COUNT = 60        # Number of LED pixels.
LED_PIN = 18          # GPIO pin connected to the pixels (18 uses PWM!).
# LED_PIN = 10        # GPIO pin connected to the pixels (10 uses SPI /dev/spidev0.0).
LED_FREQ_HZ = 800000  # LED signal frequency in hertz (usually 800khz)
LED_DMA = 10          # DMA channel to use for generating a signal (try 10)
# LED_BRIGHTNESS = 255   # Set to 0 for darkest and 255 for brightest
LED_BRIGHTNESS = 51   # Set to 0 for darkest and 255 for brightest
LED_INVERT = False     # True to invert the signal (when using NPN transistor level shift)
LED_CHANNEL = 0       # set to '1' for GPIOs 13, 19, 41, 45 or 53

# Create a NeoPixel object with the appropriate configuration.
strip = PixelStrip(LED_COUNT, LED_PIN, LED_FREQ_HZ, LED_DMA, LED_INVERT, LED_BRIGHTNESS, LED_CHANNEL)

# Initialize the library (must be called once before other functions).
strip.begin()
print("Test beginning ✅")

def clear_buffer():
  for i in range(LED_COUNT):
    strip.setPixelColor(i, Color(0, 0, 0))
  strip.show()
  sleep(1.0)

def rgb_test():
  # print("strip.numPixels() =", strip.numPixels())
  while True:
    for i in range(strip.numPixels()):
      # red led ✅
      strip.setPixelColor(i, Color(255, 0, 0))
      strip.show()
      sleep(0.1)
    sleep(1.0)
    for i in range(strip.numPixels()):
      # green led ✅
      strip.setPixelColor(i, Color(0, 255, 0))
      strip.show()
      sleep(0.1)
    sleep(1.0)
    for i in range(strip.numPixels()):
      # blue led ✅
      strip.setPixelColor(i, Color(0, 0, 255))
      strip.show()
      sleep(0.1)
    sleep(1.0)

try:
  rgb_test()
except KeyboardInterrupt:
  print('Ctrl + C exit ✅')
  clear_buffer()
  sleep(2.0)
except RuntimeError as error:
  print("error =", error, error.args[0])
  clear_buffer()
  pass
except Exception as error:
  print("exception =", error)
  clear_buffer()
  raise error
finally:
  print("clear buffer ✅")
  clear_buffer()
  sleep(3.0)
"""
$ sudo ./pixel-strip-rgb.py
"""
```

[image]