rpi-ws281x/rpi-ws281x-python #99: Unable to pip install on macos

IanWhalen, 2023-08-31:
Similar to #12 , has anyone else run into trouble installing this library on macos?  M1 chip with macos 13.5.1.

I'm trying to my development locally and then actually run on a raspberry pi, but running `poetry add rpi-ws281x` gives me:

```shell
   × Building wheel for rpi-ws281x (pyproject.toml) did not run successfully.
    │ exit code: 1
    ╰─> [20 lines of output]
        running bdist_wheel
        running build
        running build_py
        Compiling ws281x library...
        creating build
        creating build/lib.macosx-13.2-arm64-cpython-311
        creating build/lib.macosx-13.2-arm64-cpython-311/rpi_ws281x
        copying rpi_ws281x/rpi_ws281x.py -> build/lib.macosx-13.2-arm64-cpython-311/rpi_ws281x
        copying rpi_ws281x/__init__.py -> build/lib.macosx-13.2-arm64-cpython-311/rpi_ws281x
        running build_ext
        building '_rpi_ws281x' extension
        creating build/temp.macosx-13.2-arm64-cpython-311
        creating build/temp.macosx-13.2-arm64-cpython-311/lib
        clang -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -I. -I/Users/ian/Library/Caches/pypoetry/virtualenvs/freenode-led-1BZ2-4XO-py3.11/include -I/Users/ian/.pyenv/versions/3.11.2/include/python3.11 -c lib/dma.c -o build/temp.macosx-13.2-arm64-cpython-311/lib/dma.o
        clang -Wsign-compare -Wunreachable-code -DNDEBUG -g -fwrapv -O3 -Wall -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -I/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include -I. -I/Users/ian/Library/Caches/pypoetry/virtualenvs/freenode-led-1BZ2-4XO-py3.11/include -I/Users/ian/.pyenv/versions/3.11.2/include/python3.11 -c lib/mailbox.c -o build/temp.macosx-13.2-arm64-cpython-311/lib/mailbox.o
        lib/mailbox.c:40:10: fatal error: 'sys/sysmacros.h' file not found
        #include <sys/sysmacros.h>
                 ^~~~~~~~~~~~~~~~~
        1 error generated.
        error: command '/usr/bin/clang' failed with exit code 1
        [end of output]
    
    note: This error originates from a subprocess, and is likely not a problem with pip.
    ERROR: Failed building wheel for rpi-ws281x
  Failed to build rpi-ws281x
```

same error with plain old `pip install rpi-ws281x`.

I'm not seeing why sysmacros.h should be missing or any indication that its a linux-specific file, so I assume I should be able to install?

Gadgetoid, 2023-10-05:
You'd need stubs or a dummy library for this. Installing on anything but a Pi is very much out of scope.

IanWhalen, 2023-10-05:
ok, thanks for the clarification.

closed 2023-10-05 by IanWhalen