rpi-ws281x/rpi-ws281x-python #106: RPi 5 support(?)

GuyPerets106, 2023-12-24:
Hi , I'm using RPi 5 after using the same ws2812b strip on the RPi 4. It now says `Hardware revision is not supported`. 
It is expected soon?
Thanks

JustBKen, 2024-10-18:
JGraff just made a rpi5 build of the library. I havent been able to get it working with the python bindings yet tho. Anyone any tipps?

JustBKen, 2024-10-21:
https://github.com/jgarff/rpi_ws281x/issues/528#issuecomment-2421021166

RPI5 support is currently WIP. It works, but not integrated in the mainline. Would be great if this fix is integrated into this repo.

Gadgetoid, 2024-10-31:
J. dropped me a heads up, and I've been watching the GitHub notifications flood in. Taking a bit of a back seat because I have *so much* else to do, but once it's in mainline I'm happy to bump the module here.

Possible we need a bunch of bugfixes to the bindings too, since the issues are piling up 🫠

jgarff, 2024-10-31:
> J. dropped me a heads up, and I've been watching the GitHub notifications flood in. Taking a bit of a back seat because I have _so much_ else to do, but once it's in mainline I'm happy to bump the [...]

In the meantime, it might be good to have a rpi5 branch that points to the rpi_ws281x C library rpi5 branch.  Then folks could build it on that branch and it would work in Python without having to do submodule/copy tricks.  I can help out there, but I don't know if I have permission to do this in the python repo.

Gadgetoid, 2024-11-01:
> In the meantime, it might be good to have a rpi5 branch that points to the rpi_ws281x C library rpi5 branch.

I did wonder, as I typed the above, if that would be a smart idea!

Gadgetoid, 2024-11-01:
Tagged a release and hung some (untested) test builds here: https://github.com/rpi-ws281x/rpi-ws281x-python/releases/tag/pi5-beta1

Hopefully it just works, all my Pi 5s are busy! 😆 😬

GuyPerets106, 2024-11-18:
> I've since tested and released a new build here: https://github.com/rpi-ws281x/rpi-ws281x-python/releases/tag/pi5-beta2

I get this error when using the `pip` command in the link you provided:
`ERROR: rpi_ws281x-6.0.0-cp311-cp311-linux_aarch64.whl is not a supported wheel on this platform.`

timonsku, 2024-11-18:
On my install (after doing a apt upgrade) I get a `RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)`
Though it is working with the C library test program. Is the python lib shipping its own copy of the library and maybe not using the Pi5 branch? Though not sure how it could have worked so far.
My Pi5 has a revision code 0xd04170

timonsku, 2024-11-18:
Nevermind this was from an older install that snuck into system packages and not passing the venv path to sudo -E correctly.

Gadgetoid, 2024-11-19:
> There is an expected time for a full release?

🤷 this is based upon a development branch of rpi_ws281x discussed here ( https://github.com/jgarff/rpi_ws281x/issues/528 ) so whenever that's merged, I guess!

> I think it's due to the fact I'm using Python 3.12, and your whl file denotes cp311.

That'll probably do it. I don't know how compatible the bindings are but you could try just renaming the file to `rpi_ws281x-6.0.0-cp312-cp312-linux_aarch64.whl` iirc (I could easily be making this up) it actually parses the Python wheel filenames.

Unfortunately I can't connect to my build Pi right now to try cutting a 312 build, but you could try building from source.

Tetrikus, 2024-11-19:
Unfortunately I'm still struggling with this installation. I've built the kernel driver, the device tree overlay, then loaded the kernel driver, as described [here](https://github.com/jgarff/rpi_ws281x/wiki/Raspberry-Pi-5-Support). The Installation went well and testing the C-Script was running as it should. After I installed the pi5-beta2 from here into the venv and it worked fine. Now, after a reboot, it stopped working again with the original message "Hardware revision is not supported".

`File "/home/pi/venvpp/lib/python3.11/site-packages/rpi_ws281x/rpi_ws281x.py", line 135, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)`

When I do all the steps again, it works again, until I reboot the device. What am I missing?

And for when it works, is there already a possibility to use multiple channels/GPIO pins for multiple strips at the same time? As far as I can see, the `LED_PIN` in the Strip Configuration of the Python script is just ignored at the moment, right?

GuyPerets106, 2024-11-20:
> That'll probably do it. I don't know how compatible the bindings are but you could try just renaming the file to `rpi_ws281x-6.0.0-cp312-cp312-linux_aarch64.whl` iirc (I could easily be making this [...]

After renaming the file it successfully installed it. But the examples (`strandtest.py` for example) won't work. 
I did everything regarding the linux-headers/device tree overlay etc. and the `test` compiled C script runs fine.

The error I get is `ModuleNotFoundError: No module named '_rpi_ws281x'`

Just to be sure - when running `sudo --preserve-env PATH python3 strandtest.py` I should replace `PATH python3` with something like **`sudo --preserve-env ~/.pyenv/versions/<MYENVNAME>/bin/python3 strandtest.py`** (I'm using pyenv for virtual evns)

Gadgetoid, 2024-11-20:
> Just to be sure - when running `sudo --preserve-env PATH python3 strandtest.py` I should replace `PATH python3` with something like **`sudo --preserve-env ~/.pyenv/versions/<MYENVNAME>/bin/python3 strandtest.py`** (I'm using pyenv for virtual evns)

@GuyPerets106  - `PATH` is the name of the environment variable to preserve, so it should just be `sudo --preserve-env PATH python3 strandtest.py` or you can `sudo /home/<USER>/.pyenv/versions/<MYENVNAME>/bin/python3 strandtest.py` which I believe has the same effect. You're preserving `PATH ` so that sudo executes the venv Python rather than the system Python.

@Tetrikus - did you re-activate the virtual environment after rebooting?

@anico97as - are you using `sudo --preserve-env PATH python3 strandtest.py` to ensure you're actually running the virtualenv version of the package and not the system package?

anico97as, 2024-11-20:
hi @Gadgetoid,
thanks for your reply. I solved that by repeating following steps:

```
    Loading the kernel module from the driver directory. Use the pwm_channel from the GPIO list above. The following example applies to GPIO 14 and 18.

        $ sudo insmod ./rp1_ws281x_pwm.ko pwm_channel=2

    Loading the device tree overlay. This will initialize the device.

        $ sudo dtoverlay -d . rp1_ws281x_pwm

    Setting up the GPIO (for GPIO 18). Use the GPIO number and aX parameter from the list above.

        $ sudo pinctrl set 18 a3 pn
```
After that it was not a problem to control my 8x32 matrix. 
However I can confirm what @Tetrikus said, after a reboot i will again get the error "Hardware revision is not supported...". Then I have to repeat above mentioned steps in order to get it working until the next reboot. 

Could it be that the device tree overlay is not persisted somehow and that it will not be loaded by the kernel by default?

GuyPerets106, 2024-11-20:
> @GuyPerets106 - `PATH` is the name of the environment variable to preserve, so it should just be `sudo --preserve-env PATH python3 strandtest.py` or you can `sudo [...]

So unfortunately it does not work, because I did just that :/ 
I get an error about the module not found, so probably I miss a step(?)
It seems that it can't find the `_rpi_ws281x` module - you know what it's related to?

Thanks for the help!

Gadgetoid, 2024-11-20:
> So unfortunately it does not work, because I did just that :/ I get an error about the module not found, so probably I miss a step(?) It seems that it can't find the `_rpi_ws281x` module - you know what it's related to?

@GuyPerets106  - Sorry too many crossed wires and I forgot to properly respond to you here. I'm pretty sure this is because the .so file that contains the `_rpi_ws281x` C part of the module is still named `_rpi_ws281x.cpython-311-aarch64-linux-gnu.so`. You could probably rename that in the same way, using `pip show rpi-ws281x` to find where it's installed.

I'll cut a proper 3.12 build in the meantime. (Raspberry Pi OS bookworm is still on 3.11, are you using another OS?)

Tetrikus, 2024-11-21:
> @Tetrikus - did you re-activate the virtual environment after rebooting?
Yes, I've re-activating it after rebooting and have the same issue every time.

Tetrikus, 2024-12-02:
Is it somehow already possible to address two different Strips with the current beta version?

lbt, 2024-12-15:
@jgarff @Gadgetoid  I'm working on another feature and tried to rebase to the pi5 branch and it failed on a PiZero2
To check for regressions I've now gone to a vanilla pi5 branch build and it too fails for a Zero2 with an SPI control on pin 10:
```
uccessfully installed rpi_ws281x-6.0.0
(venv-leds) pi@lichee:/everything/devel/raspi/led/rpi-ws281x-python/library $ python ../examples/strandtest.py Press Ctrl-C to quit.
Use "-c" argument to clear LEDs on exit
Color wipe animations.
Can't send spi messageTraceback (most recent call last):
  File "/everything/devel/raspi/led/rpi-ws281x-python/library/../examples/strandtest.py", line 107, in <module>
    colorWipe(strip, Color(255, 0, 0))  # Red wipe
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/everything/devel/raspi/led/rpi-ws281x-python/library/../examples/strandtest.py", line 28, in colorWipe
    strip.show()
  File "/everything/devel/raspi/led/rpi-ws281x-python/library/rpi_ws281x/rpi_ws281x.py", line 142, in show
    raise RuntimeError('ws2811_render failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_render failed with code -14 (SPI transfer error)
Segmentation fault
```

Doing a git checkout 1f47b59 on the lib submodule (and redoing the venv pip install to rebuild the .so libs) results in a a working test whilst still on the pi5-beta2 branch of rpi-ws281x-python

Just so you know that this causes problems for existing SPI-based solutions which used to be the most CPU friendly for lower powered devices.

jxns, 2025-02-06:
Hey, just wanted to ask for a quick status update on this. Is there already a working solution that I could use for Pi 5?

wenbo-y, 2025-04-10:
> Hey, just wanted to ask for a quick status update on this. Is there already a working solution that I could use for Pi 5?

https://github.com/jgarff/rpi_ws281x/issues/528  , lastest comment