rpi-ws281x/rpi-ws281x-python #111: RPi 3 Model A+ Rev 1.1 - Hardware revision is not supported

sydios, 2024-04-18:
Hello,

I got the new RPi 3 Model A+ Rev 1.1. And I get the error that the hardware is not supported. 
Is there a setting I can change to make this work?

Thanks!

`File "/usr/local/lib/python3.11/dist-packages/rpi_ws281x/rpi_ws281x.py", line 143, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -3 (Hardware revision is not supported)
Segmentation fault
`
`Revision        : 9020e1

Serial          : 0000000095ced7a7

Model           : Raspberry Pi 3 Model A Plus Rev 1.1
`

sjewula, 2024-07-28:
Hello,

I just received the same Pi 3 Model A+ Rev 1.1 and I am having the same problem.
@sydios did you ever find a solution?

sydios, 2024-07-28:
@sjewula yes. This one worked: https://github.com/jgarff/rpi_ws281x/issues/483

sjewula, 2024-07-29:
> @sjewula yes. This one worked: [jgarff/rpi_ws281x#483](https://github.com/jgarff/rpi_ws281x/issues/483)

@sydios Thank you for the link.  I'm doing something wrong.  I still cannot get this to work.