rpi-ws281x/rpi-ws281x-python #116: ModulNotFound Error

JThielenSchool, 2024-11-02:
I wanted to install the module on my Raspberry Zero in a virtual environment.

It works fine with pip, the module is found in python. Pip installs it in the correct environment.

But as soon as I run the strandtest.py, I get the error ModulNotFound.

Without virtual environment with compiling the c-files by myself it works...

Gadgetoid, 2024-11-04:
The little cogs in my brain just unstuck; are you running with `sudo`?

JThielenSchool, 2024-11-05:
ok my fault, 

when i start the programm with sudo i leave the virtual environment, so he can't find it, if the modul is only install in the environment, therefore i have to install it on the global python-path.

And we need sudo to get access to the pins, right? I can't start it, without sudo.

Gadgetoid, 2024-11-05:
Use `sudo --preserve-env PATH python example.py` and it *should* work. I've published a new version of the package, too, which fixes some bugs and documents this caveat.

JThielenSchool, 2024-11-05:
Yeah thank you, that works for me.

closed 2024-11-05 by JThielenSchool